legokor/INDACT_ROBOTKAR #16: Getting started with Ardunio Nano #3

oprezz, 2019-09-17:
Az eszköz egy Arduino Nano lesz + HW az egyes feladatoknál szóba jöhet, ezeket megtaláljátok a laborban.

Hogyan is álljatok neki?
- [ ] Az [Arduino Nano](https://store.arduino.cc/arduino-nano) egy 8bites ATMega328 mikroprocesszorra épül, telis tele perifériákkal. A fenti oldalon nézz körül, olvasd el a technikai adatlapját, dokumentációkat. Ez röviden leírja a legfontosabb tudnivalókat (ha valami nem tiszta kérdezz!).

- [ ] Töltsd le a [fejlesztőkörnyezetet](https://www.arduino.cc/en/Guide/ArduinoNano), ha még nem tetted meg (ne a webeset, hanem a desktop IDE-t), és fordíts rá egy példafeladatot. Tanulmányozd picit a kódot :)

Nyisd meg, értsd meg, nyúlj bele, fordítsd le minimum az alábbi példaprogramokat:
Basics:
- [ ] LED blinking 
- [ ] AnalogReadSerial
- [ ] DigitalReadSerial
- [ ] ReadAnalogVoltage
- [ ] Button
- [ ] Debounce
- [ ] AnalogInput

Advanced:
- [ ] Servo // Knob, Sweep
- [ ] LED villogtatása HW-es Timerrel
- [ ] PWM jel előállítása

Természetesen a feladat során használjatok library-ket, akár neteseket, akár "beépítettet" (erről később).

_bonusz_
egy-két videó, ha nem tudnátok elaludni :)
[Learn Arduino in 15 mins](https://www.youtube.com/watch?v=nL34zDTPkcs)
[Tutorial 01: Hardware Overview: Arduino Course for Absolute Beginners (ReM)](https://www.youtube.com/watch?v=09zfRaLEasY)

Szibence, 2019-09-18:
Letöltöttem a fejlesztőkörnyezetet, vele együtt a drivereket. Egy lego laborban talált mini USB kábellel csatlakoztattam az Arduino-t a laptopomhoz, mire azt a windows 10 nem ismerte fel. Energiát láttam, hogy kapott, mivel a power led világítótt, azonban a gépen a device manager sem mutatta, hogy lenne bármi is a porton. Egy ideig eltartó hibakeresés újraindítgatás és ki-be kötögetés után tudtam szerezni egy másik kábelt, amivel tökéletesen működik. Sose bízzatok elsőre egy kábelben.

oprezz, 2019-09-18:
Fincsi bug, azon kívül, hogy nem volt jó a kábel, egy olyan tippem lenne, hogy lehet csak "töltő" kábel volt. 
[image]
Ilyenkor az USB 4 pinjéből csak 2-t használnak (ált olyan eszközhöz adják, aminél az USB csak tölt, adatot nem lehet rajta keresztül másolni) olcsósítás céljából. De a következtetés nagyon is helyes :D

Szibence, 2019-09-20:
LED blinking:
Kódban két fő függvény:
setup() - benne inicializáló függvények
- pinMode
loop() - végtelen ciklus a programkóddal
- digitalWrite
- delay

closed 2020-11-14 by Szibence